syssi/esphome-dps #14: DPS Model DPS5005 shows Amps with a multiplication factor of 10

hellvideorazor, 2023-07-02:
Watt display is OK bit Amp displays with a factor of 10. Therefore limited to max 20 = 2A.
[image]

syssi, 2023-07-02:
That's interesting. Did you use the `number` entity to set the output current? Did you set 20 A or 2 A using the number entity? I would like to understand two or three entities should be fixed for your device model / firmware version:

```
sensor.dps_output_current
sensor.dps_current_setting
number.dps_current_setting
```

hellvideorazor, 2023-07-02:
Its the same wherever I set current. in homeassistant or on the device. If I set 4A on the device it shows 40A in the integration. If I set 20A in Homeassistant it shows 2A on the DPS.
`[23:29:33][D][sensor:094]: 'dps voltage setting': Sending state 6.02000 V with 2 decimals of accuracy
[23:29:33][D][number:012]: 'dps voltage setting': Sending state 6.020000
[23:29:33][D][sensor:094]: 'dps current setting': Sending state 20.00000 A with 2 decimals of accuracy
[23:29:33][D][number:012]: 'dps current setting': Sending state 20.000000
[23:29:33][D][sensor:094]: 'dps output voltage': Sending state 2.74000 V with 2 decimals of accuracy
[23:29:33][D][sensor:094]: 'dps output current': Sending state 19.98000 A with 2 decimals of accuracy
[23:29:33][D][sensor:094]: 'dps output power': Sending state 5.47000 W with 2 decimals of accuracy
[23:29:33][D][sensor:094]: 'dps input voltage': Sending state 18.61000 V with 2 decimals of accuracy
[23:29:34][D][text_sensor:064]: 'dps protection status': Sending state 'Normal'
[23:29:34][D][sensor:094]: 'dps backlight brightness': Sending state 40.00000 % with 0 decimals of accuracy
[23:29:34][D][text_sensor:064]: 'dps device model': Sending state 'DPS5005'
[23:29:34][D][sensor:094]: 'dps firmware version': Sending state 1.40000  with 1 decimals of accuracy`

syssi, 2023-07-03:
I will prepare a fix!

hellvideorazor, 2023-07-04:
I installed the branch today. Same issue.
[06:48:01][D][sensor:094]: 'dps voltage setting': Sending state 6.23000 V with 2 decimals of accuracy
[06:48:01][D][number:012]: 'dps voltage setting': Sending state 6.230000
[06:48:01][D][sensor:094]: 'dps current setting': Sending state 25.00000 A with 2 decimals of accuracy
[06:48:01][D][number:012]: 'dps current setting': Sending state 25.000000
[06:48:01][D][sensor:094]: 'dps output voltage': Sending state 3.35000 V with 2 decimals of accuracy
[06:48:01][D][sensor:094]: 'dps output current': Sending state 24.92000 A with 2 decimals of accuracy
[06:48:01][D][sensor:094]: 'dps output power': Sending state 8.34000 W with 2 decimals of accuracy
[06:48:01][D][sensor:094]: 'dps input voltage': Sending state 18.60000 V with 2 decimals of accuracy
[06:48:01][D][text_sensor:064]: 'dps protection status': Sending state 'Normal'
[06:48:01][D][sensor:094]: 'dps backlight brightness': Sending state 40.00000 % with 0 decimals of accuracy
[06:48:01][D][text_sensor:064]: 'dps device model': Sending state 'DPS5005'
[06:48:01][D][sensor:094]: 'dps firmware version': Sending state 1.40000  with 1 decimals of accuracy

syssi, 2023-07-04:
This was too fast. The branch doesn't provide the fix yet. I will ping you as soon it's ready. :-)

syssi, 2023-07-04:
Please update the `dps` section of your YAML to:

```
dps:
  id: dps0
  modbus_id: modbus0
  update_interval: 5s
  current_resolution: HIGH
```

And try the feature branch another time.

hellvideorazor, 2023-07-04:
Failed config dps: [source /config/esphome/dps1.yaml:61] - id: dps0 modbus_id:
modbus0 update_interval: 5s [current_resolution] is an invalid option for
[0]. Please check the indentation. current_resolution: HIGH

um 07:48:

syssi, 2023-07-04:
It looks like you aren't using the feature branch. Please change to branch of the external component from `main` to `make-resolution-configurable`: 

```
substitutions:
  name: dps
  device_description: "Monitor and control a RDTech Digital Control Power Supply (DPS) via UART-TTL"
  external_components_source: github://syssi/esphome-dps@make-resolution-configurable
```

hellvideorazor, 2023-07-04:
All set. Even uploaded and tried config change afterwards again. I will
check tomorrow again - on mobile & vpn for today

um 08:14:

hellvideorazor, 2023-07-04:
My bad on config. Works like a charm!
[image]

closed 2023-07-04 by hellvideorazor

syssi, 2023-07-04:
Did you try to use the number entity to change the current?

hellvideorazor, 2023-07-04:
Yes, I did. works now with 5a but gets up to 4.89

um 19:23:

syssi, 2023-07-04:
I've merged the new feature into main. Please update your YAML again.

hellvideorazor, 2023-07-05:
updated & still working. Thanks a lot for the quick response and assistance!

syssi, 2023-07-05:
You are welcome! Feel free to create new issues if you miss something or there is room for improvements!