syssi/esphome-dps #20: Add XY6020 support

syssi, 2023-11-05:
https://github.com/kbs3056/XY6015L_pyGUI
https://github.com/syssi/esphome-dps/discussions/19#discussioncomment-7357252

Baud rate: 115200

```
[18:30:22][I][dps:063]: Status frame received
[18:30:22][D][text_sensor:064]: 'dps device model': Sending state 'DPS21'
[18:30:22][D][sensor:094]: 'dps voltage setting': Sending state 25.00000 V with 2 decimals of accuracy
[18:30:22][D][number:012]: 'dps voltage setting': Sending state 25.000000
[18:30:22][D][sensor:094]: 'dps current setting': Sending state 0.20900 A with 3 decimals of accuracy
[18:30:22][D][number:012]: 'dps current setting': Sending state 0.209000
[18:30:22][D][sensor:094]: 'dps output voltage': Sending state 25.00000 V with 2 decimals of accuracy
[18:30:22][D][sensor:094]: 'dps output current': Sending state 0.00100 A with 3 decimals of accuracy
[18:30:22][D][sensor:094]: 'dps output power': Sending state 0.02000 W with 2 decimals of accuracy
[18:30:22][D][sensor:094]: 'dps input voltage': Sending state 48.26000 V with 2 decimals of accuracy
[18:30:22][D][text_sensor:064]: 'dps protection status': Sending state 'Normal'
[18:30:22][D][sensor:094]: 'dps backlight brightness': Sending state 20.00000 % with 0 decimals of accuracy
[18:30:22][D][sensor:094]: 'dps firmware version': Sending state 1.50000 with 1 decimals of accuracy
[18:30:22][W][component:214]: Component modbus took a long time for an operation (0.14 s).
[18:30:22][W][component:215]: Components should block for at most 20-30ms.
[18:30:22][D][uart_debug:158]: <<< "\x01\x03\x1A\t\xC4\x00\xD1\t\xC4\x00\x01\x00\x02\x12\xDA\x00\f\x00\x00\x00\xFA\x00\x00\x00\x01\x00\x15\x00\x0F\x91}"
```

Not working:
- Output switch
- Firmware version

syssi, 2023-11-06:
Fork: https://github.com/creepystefan/esphome-XY6020

creepystefan, 2024-06-02:
https://github.com/creepystefan/ESPhome-Sinilink-WF-POW

hey 
also mein xy6020 Batterie Lader funktioniert sehr gut.
danke nochmals für das anschuppsen des Projektes.
habe dir noch einen Link geschickt für eine nette spielerei, falls es dich interresiert.

Ist das Original Wifi Modul von Sinilink für das XY6020
Grüße

syssi, 2024-06-02:
Cool!

Tiese, 2024-06-11:
Hallo @creepystefan,

kann das WiFi-Modul über Web-API oder direkt über URL abgefragt werden?
Oder muss man die Sinilink-App benutzen?

Vielen Dank für deine Hilfe!

Schöne Grüße
Tiese

creepystefan, 2024-06-11:
Hey @Tiese 
Das sinilink wifi Modul kann wie ein normaler ESP benutzt werden.
Am besten schaust du auf meiner Seite. Dort habe ich die GPIOs
Und programmieranschlüsse angezeigt.
In meinem Beispiel wird das Modul mit ESPHome bespielt.
Und kann alles gemacht werden.
Allerdings muss ich noch den Code für das sinilink mit dem XY6020 anpassen. Momentan ist es für einen ESP32 angepasst.
Bitte schreibe mir doch direkt auf meiner GitHub Seite.
API wird über ESPhome benutzt
Auch eine Web Oberfläche.
Das Sinilink Portal wird nicht gebraucht.

syssi, 2024-09-30:
Stefan hat hier den gesamten Prozess erklaert: https://github.com/creepystefan/esphome-XY6020

Magst du in seinem Projekt einen `Issue` eroeffnen und erlaeutern, an welcher Stelle du Hilfe benoetigst?

creepystefan, 2024-09-30:
hey
klar kannst du bei mir ein Issue öffnen. ich kann dir gerne weiter helfen.
ich habe es genau so am laufen bei mir.
Bei Übereinspeisung wird der Strom am XY6020 über den WFPOW geregelt
und somit eine Batterie geladen.
Ich müsste es mal als Bild zeichnen und einstellen.
WFPOW ist ein spezielle ESP8285 Form mit Steckverbindung zum XY6020.
wird ja mit dem XY6020 als -W Version mitgeliefert.

roseweau, 2024-09-30:
Danke,  bin schon in Kontakt mit Stefan.--Gesendet mit der WEB.DE Mail App

roseweau, 2024-09-30:
Hallo Stefan, vielen Dank für die prompte Antwort. Ich kann gerne einen Issue formulieren,  aber prinzipiell verstehe ich (noch) nicht, wie ich mich mit dem XY-FWPOW also dem ESP8285 verbinden um dann ggf. Änderungen an Strom oder Spannung am XY-6020L zu machen. In diversen Youtubes ( polnisch oder koreanisch) sehe ich eine Kopplung mit einem IPhone mit Touch-button, ich hab Android. Irgendwo steht was von AP Mode (access point WiFi) aber ich komm da nicht weiter.
 In github sehe ich einen Arduino, den könnte ich ja durch einen ESP32 ersetzen und analog zum ATMega via UART kommunizieren. Aber mich würde der 8285 auch interessieren,  ich hab ihn ja
Gruß Werner --Gesendet mit der WEB.DE Mail App

creepystefan, 2024-09-30:
So einfach ist das nicht.
Auf dem WFPow (esp8285) läuft die standart Firmware des Herstellers.
Diese muss man sozusagen löschen und den ESP neu programmieren mit esphome. Das fungiert als wlan Schnittstelle.
Über die Originale Firmware geht es nicht!
Schau dir mal an wie ESPhome funktioniert.
Und bitte öffne bei mir einen Issue.
Über ESPhome kann man dann über RS232 Modbus das Netzteil steuern.