syssi/esphome-dps #21: DPS3005 no response

CommanderRedYT, 2023-12-06:
[image]

I have connected a ESP32 (NodeMCU-32 V1.3) to a DPS3005. Mine came without a connector, but i just soldered some cables to it. 

I double-checked all connections, tried to switch RX and TX, tried different Pins but still, it does not work...

syssi, 2023-12-07:
Which GPIOs did you use as RX/TX pins? Could you provide a photo of your wiring and the configuration YAML?

CommanderRedYT, 2023-12-07:
I wired it like this: https://johan.kanflo.com/wp-content/uploads/2017/06/DPS5005-comms-UART.jpg (Not my picture, but is same as i have)

CommanderRedYT, 2023-12-10:
Okay, somehow it is working now. Did not change anything tho.

closed 2023-12-10 by CommanderRedYT