syssi/esphome-dps #23: Alternatve Firmware Profimax 4.3 issue

btmerz, 2024-04-07:
Hello,
I have installed the DPS firmware versin 4.3 from Profimax (very good firmware for this device). So I testet it with this esphome software. After changing the uart speed to 115200 I was able to communicate with my power supply. 
Unfortunately I am only able to read the values, but I can't write anything.
There is a ESP01 installed. When I use my own Arduino firmware everything works fine. So the wiring is correct.

Do someone have any advice?
Tanks in advance.
Bernd

syssi, 2024-04-07:
Could you provide some details about the profimax firmware? You could enable the debug output of the `uart` component to dump the RAW traffic (requests) at the log: https://github.com/syssi/esphome-dps?tab=readme-ov-file#debugging

Please compare the RAW traffic (write commands) with the payloads of your arduino implementation. Is there something different?

btmerz, 2024-04-08:
Hello Sebastian, for now I can provide you a link to the firmware documentation. There is also a chapter about the modbus protocol...https://profimaxblog.ru/dps_update/Best regards Bernd Gesendet von Outlook für Android

btmerz, 2024-04-09:
Hello Sebastian, 
here ist the log output, when I try to switch on the power:

22:44:05 | [D] | [uart_debug:114] | <<< 01:03:1A:02:D8:00:64:00:01:00:00:00:00:04:70:00:00:00:00:00:00:00:00:00:06:13:8D:00:2B:85:C2
22:44:06 | [D] | [switch:012] | 'dps-5005 output' Turning ON.
22:44:06 | [D] | [uart_debug:114] | >>> 01:10:00:09:00:01:02:00:01:67:09
22:44:09 | [D] | [uart_debug:114] | >>> 01:03:00:00:00:0D:84:0F
22:44:09 | [I] | [dps:063] | Status frame received
22:44:09 | [D] | [text_sensor:064] | 'dps-5005 device model': Sending state 'DPS5005'
22:44:09 | [D] | [sensor:093] | 'dps-5005 voltage setting': Sending state 7.28000 V with 2 decimals of accuracy
22:44:09 | [D] | [number:012] | 'dps-5005 voltage setting': Sending state 7.280000
22:44:09 | [D] | [sensor:093] | 'dps-5005 current setting': Sending state 0.10000 A with 3 decimals of accuracy
22:44:09 | [D] | [number:012] | 'dps-5005 current setting': Sending state 0.100000
22:44:09 | [D] | [sensor:093] | 'dps-5005 output voltage': Sending state 0.01000 V with 2 decimals of accuracy
22:44:09 | [D] | [sensor:093] | 'dps-5005 output current': Sending state 0.00000 A with 3 decimals of accuracy
22:44:09 | [D] | [sensor:093] | 'dps-5005 output power': Sending state 0.00000 W with 2 decimals of accuracy
22:44:10 | [D] | [sensor:093] | 'dps-5005 input voltage': Sending state 11.35000 V with 2 decimals of accuracy
22:44:10 | [D] | [text_sensor:064] | 'dps-5005 protection status': Sending state 'Normal'
22:44:10 | [D] | [sensor:093] | 'dps-5005 backlight brightness': Sending state 120.00000 % with 0 decimals of accuracy
22:44:10 | [D] | [sensor:093] | 'dps-5005 firmware version': Sending state 4.30000  with 1 decimals of accuracy
22:44:10 | [W] | [component:214] | Component modbus took a long time for an operation (0.12 s).
22:44:10 | [W] | [component:215] | Components should block for at most 20-30ms.
22:44:10 | [D] | [uart_debug:114] | <<< 01:03:1A:02:D8:00:64:00:01:00:00:00:00:04:6F:00:00:00:00:00:00:00:00:00:06:13:8D:00:2B:DA:CA
22:44:14 | [D] | [uart_debug:114] | >>> 01:03:00:00:00:0D:84:0F
22:44:14 | [I] | [dps:063] | Status frame received

My arduino code to switch on the power supply is as follows:

#include <ModbusMaster.h>
Serial.begin(115200);
node.begin(1, Serial);
node.writeSingleRegister(9, EinAus);  // Übergabe EinAus an DPS5005 (Register 9)

I hope this info is helpful.
Bernd

btmerz, 2024-04-27:
Hello, do you have any idea what could be the problem?

syssi, 2024-04-29:
Please change this line:

```
substitutions:
  external_components_source: github://syssi/esphome-dps@main
```

to

```
substitutions:
  external_components_source: github://syssi/esphome-dps@test-write-single-register
```

and recompile + flash the ESP. Does it work now?

btmerz, 2024-04-30:
Hello Sebastian,
I've tested the new version. unfortunately I still can't controll my DPS. 
Here is the log file:

23:05:53 | [D] | [uart_debug:114] | <<< 01:03:1A:03:28:00:64:00:00:00:00:00:00:04:6E:00:00:00:00:00:00:00:00:00:06:13:8D:00:2B:F5:24
-- | -- | -- | --
23:05:54 | [D] | [switch:012] | 'dps-5005 output' Turning ON.
23:05:55 | [D] | [uart_debug:114] | >>> 01:06:00:09:00:01:98:08
23:05:56 | [D] | [uart_debug:114] | >>> 01:03:00:00:00:0D:84:0F
23:05:56 | [I] | [dps:063] | Status frame received

...

syssi, 2024-05-01:
The payload of the feature branch is equal to the payload of `node.writeSingleRegister(9, EinAus);`. I've no idea why it's not working.

```
# implemented
Output on:   01:10:00:09:00:01:02:00:01:67:09
Output off:  01:10:00:09:00:01:02:00:00:a6:c9

# feature branch
Output on:   01:06:00:09:00:01:98:08
Output off:  01:06:00:09:00:00:59:C8
``` 

Profimax Modbus registers: https://profimaxblog.ru/modbus-communication-protocol-2/

btmerz, 2024-07-07:
Meanwhile I organized a logic analyzer to campare the incomming signals.
Here is the sending sequence to switch on the DPS comming from my arduino code:
[image]

When I do the same with your ESPHome test-write-single-register code I got the following result:
[image]

There is a lot of stuff going on. The last sequence seems to be correct, but it will be somehow ignored...
[image]

As you can see also the logic analyzer software isn't able to parse the data correctly.

Hopefully this info helps you to make your code working.
Thanks in advance for your asistance

syssi, 2024-07-08:
Could you provide your YAML? I would like to replicate your setup.

btmerz, 2024-07-08:
Here is my YML file:
[DPS5005.zip](https://github.com/user-attachments/files/16128594/DPS5005.zip)

btmerz, 2024-07-31:
Hi, do you have any news on this topic?

syssi, 2024-07-31:
Please set the `baud_rate` of the `logger` component to `0`. At the moment you are logging to `UART0` too which causes a lot of noise. With UART logging disabled the UART signal of your YAML looks good:

[image]
[image]

btmerz, 2024-07-31:
THANK YOU VERY MUCH!!!!
That did the trick! Now it works as expected.
I am so happy. Now I can start my project.

All best to you. Bernd

syssi, 2024-07-31:
@btmerz Could you verify the `main` branch still doesn't work for you? So the single register write is required for your device firmware?

btmerz, 2024-08-01:
I can confirm that the main version works as well!

closed 2024-08-01 by syssi