syssi/esphome-dps #38: Set new current from within lambda

btmerz, 2025-04-11:
Hello,
currently I write a code to charge my battery from excess PV power.
Therefor I wrote a PID controller which calculates the necessary current in a lambda function. So far so good.

Now I want to send this value to the dps .
For a brief second the calculated current value appears in HA, but isn't sent to the dps, and its overwritten by the set current.
When I set a new current in HA everything works fine.

Here is the part of my code:

number:
  - platform: dps
    current_setting:
      name: "${name} current setting"
      id: set_current

#doing some calculation...
  lambda:  id(set_current).publish_state(new_current);

For me it seems there is something wrong with the way I want to set the new_current.

Could you give me some advice?
Thanks.

syssi, 2025-04-11:
Publishing isn't the right choice here. You want to control the number entity from the lambda function:

```
auto call = id(set_current).make_call();
call.set_value(new_current);
call.perform();
```

https://esphome.io/components/number/index.html#lambda-calls

btmerz, 2025-04-11:
Thank you for the assistance. I will test it tomorrow.

btmerz, 2025-04-12:
Great,  now it works. Thanks for your help 👍.

closed 2025-04-12 by syssi